meekys/cam_overlay #1: Request: Fix picture shift

maskaz, 2018-08-06:
Would be possible to fix picture shift? Picture has bigger height than screen and bottom is cutted.
(Great feature will be option to configure picture position, height and width)

Regards

meekys, 2018-08-11:
Currently there's no configuration for this, but if you edit the x/y values in **screenQuad** and **overlayQuad** in **cam_overlay.c** you should be able to adjust the position on screen.

https://github.com/meekys/cam_overlay/blob/master/cam_overlay.c#L264

Values are x, y, z, u, v for each line and the x/y values range from -1.0 for far left/top to 1.0 for far right/bottom

ie. This should display an image half the size in the middle of the screen
static GLfloat screenQuad[4][5] = { // x, y, z, u, v
    {**-0.5f,  0.5f**, 0.0f,    0.0f, 0.0f}, // Top left
    {**-0.5f, -0.5f**, 0.0f,    0.0f, 1.0f}, // Bottom left
    { **0.5f,  0.5f**, 0.0f,    1.0f, 0.0f}, // Top right
    { **0.5f, -0.5f,** 0.0f,    1.0f, 1.0f}  // Bottom right
};

After these changes you will need to re-run **make**

maskaz, 2018-08-11:
Thanks I had tried this before and by change resolution of input but problem is somewhere else.
Picture looks like stretched to 4:3 format and bottom part is cutted from picture.
Ratio change in the code did not help too.

meekys, 2018-08-12:
What resolutions are shown on startup for the cam/display?
```
Dimensions: **720 x 480** Pixel Format: YUYV
Opened screen 0 @ **1920 x 1200**
```

What resolutions are available?
`**v4l2-ctl --list-formats-ext**`

Does capturing an image using fswebcam or any other v4l2 capable software you've got installed also crop the image?
`**fswebcam --no-banner -S 100 image.jpg**`

You might also be able to fiddle with this line to adjust the display ratio, but it sounds like you've already tried this? By default it should stretch the image to the screen dimensions. If you set this to 1, it should appear as a square image.
https://github.com/meekys/cam_overlay/blob/master/cam_overlay.c#L290