meekys/cam_overlay #2: Request: Invert video (180 rotate)

abibias, 2018-10-03:
Would be possible to rotate video (180 degrees)? 

Regards

meekys, 2018-10-04:
Good idea. I'll have to add this to the list of features to add.

Until then, adjusting the x/y coordinates of screenQuad should do what you're after
https://github.com/meekys/cam_overlay/blob/master/cam_overlay.c#L264

ie. (Untested)
Rotate:
`
static GLfloat screenQuad[4][5] = { // x, y, z, u, v
    { 1.0f, -1.0f, 0.0f,    0.0f, 0.0f}, // Top left
    { 1.0f,  1.0f, 0.0f,    0.0f, 1.0f}, // Bottom left
    {-1.0f, -1.0f, 0.0f,    1.0f, 0.0f}, // Top right
    {-1.0f,  1.0f, 0.0f,    1.0f, 1.0f}  // Bottom right
};
`

Flip:
`
static GLfloat screenQuad[4][5] = { // x, y, z, u, v
    {-1.0f, -1.0f, 0.0f,    0.0f, 0.0f}, // Top left
    {-1.0f,  1.0f, 0.0f,    0.0f, 1.0f}, // Bottom left
    { 1.0f, -1.0f, 0.0f,    1.0f, 0.0f}, // Top right
    { 1.0f,  1.0f, 0.0f,    1.0f, 1.0f}  // Bottom right
};
`

abibias, 2018-10-05:
Thanks  a lot!
Works perfect! ;)

closed 2018-10-05 by abibias

TheFuzz4, 2018-10-24:
Will this work for mirroring as well?  Right now I need to have my image mirror flipped.

meekys, 2018-10-25:
Latest commit should allow you to flip horizontally, vertically, and also rotate. Hopefully that should cover all the scenarios you'll need.