meekys/cam_overlay #3: Failed to add service

TheFuzz4, 2018-10-21:
Good evening,

First off thank you for this project as I am certain this will make my OpenAuto experience just amazing.  I purchased a USB camera to use with my pi.  I've tested this camera out on my regular Ubuntu box and confirmed it does play well with linux.  I've also tested it out with vlc on the pi as well and all is good.  However when I fire off the command to start the cam_overlay it throws the following error.  I've tried it both as pi and also as root with the same results.  I also attempted to compile it as myself and as root as well with the same results.  I'm not sure what service its attempting to add here.  Thank you very much for your help and for writing the perfect reverse cam app. 

`Opened device '/dev/video0'
Dimensions: 800 x 600 Pixel Format: YUYV
* failed to add service - already in use?
`

TheFuzz4, 2018-10-21:
BTW I solved this tonight.  It turns out that I had the Open GL enabled.  I'm using OpenAuto Pro.  I disabled the OpenGL and all is right in the world.

closed 2018-10-21 by TheFuzz4

andreimoraru, 2018-12-02:
@TheFuzz4 I confirm that disabling OpenGL fixed the error for me as well.

OpenAuto Pro 2.0
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux

Steps to fix:
1) Run **raspi-config** in Raspbian shell.
2) Go to **Advanced Options**
3) Go to **A7 GL Driver**
4) Select **G3 Legacy Original non-GL desktop driver.**
5) Choose **Ok** then **Finish**. 
A reboot will be required.

Next, run ./cam_overlay with appropriate parameters for your setup.

egisz, 2020-07-29:
Hi, 
I get same error even with disabled OpenGL, running RPI4.
What rpi do you use? 
What branch: master or cpp?