meekys/cam_overlay #7: Video grabber with profile YUVY not YUYV

hawkeyexp, 2019-01-30:
Hi,

i adapted your code for crankshaft (pi based android auto headunit) and noticed a prob with a Syntek STK1160 video bridge:
the request profile is not YUYV - it is YUVY. With a easy copy to test it works with wrong colors.
Can you possible add this profile ?

[image]
[image]

If you are interested on our free project:
getcrankshaft.com
github.com/opencardev

Thx, Hawkeye

meekys, 2019-02-02:
Hi @hawkeyexp,
I've been keeping an eye on Crankshaft. The NG screenshots are looking awesome and I'm looking forward to giving it a try when I get some time to tear apart my current setup.

In regards to changing the shader from YUYV to YUVY, this should be relatively straight forwards, but I don't have the hardware to test this.
If I make copy the YUYV shader as you already have and modify the following lines, this may give what you're after.

https://github.com/meekys/cam_overlay/blob/182c1b67a133d78a5ff7d71385037a9eb03633f6/shader-YUYV.frag#L55

```
    // Y1, U , V, Y2
    maxfragp vec4 yuyv = texture2D( texture, uv );

    maxfragp vec3 yuv;
    if( fOdd <= 0.5 )
        yuv = yuyv.rgb; // Y1, U, V
    else
        yuv = yuyv.agb; // Y2, U, V
```

Once it's tested against some real hardware I'm more than happy to commit the shader, or accept a pull request

meekys, 2019-02-02:
@jonlowe I've created a new issue for your report, as I believe it to be a different issue

andreimoraru, 2019-02-27:
Hi @meekys ,

Could you please also add the shader for MJPG. My device only supports MJPG pixel format. I would also appreciate any guidance on how to create the shader. Thanks
```bash
pi@raspberrypi:~/cam_overlay $ ./cam_overlay.bin
Opened device '/dev/video0'
Dimensions: 640 x 480 Pixel Format: MJPG
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
Opened screen 0 @ 800 x 480
Cannot open 'shader-MJPG.frag': 2, No such file or directory
```
```shell
pi@raspberrypi:~/cam_overlay $ v4l2-ctl --all
Driver Info (not using libv4l2):
        Driver name   : uvcvideo
        Card type     : USB2.0 PC CAMERA
        Bus info      : usb-3f980000.usb-1.5
        Driver version: 4.14.79
        Capabilities  : 0x84200001
                Video Capture
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps   : 0x04200001
                Video Capture
                Streaming
                Extended Pix Format
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
        Width/Height      : 640/480
        Pixel Format      : 'MJPG'
        Field             : None
        Bytes per Line    : 0
        Size Image        : 614400
        Colorspace        : Default
        Transfer Function : Default
        YCbCr/HSV Encoding: Default
        Quantization      : Default
        Flags             :
Crop Capability Video Capture:
        Bounds      : Left 0, Top 0, Width 640, Height 480
        Default     : Left 0, Top 0, Width 640, Height 480
        Pixel Aspect: 1/1
Selection: crop_default, Left 0, Top 0, Width 640, Height 480
Selection: crop_bounds, Left 0, Top 0, Width 640, Height 480
Streaming Parameters Video Capture:
        Capabilities     : timeperframe
        Frames per second: 60.000 (60/1)
        Read buffers     : 0
                     brightness (int)    : min=0 max=255 step=1 default=-8193 value=128
                       contrast (int)    : min=0 max=255 step=1 default=57343 value=128
                     saturation (int)    : min=0 max=255 step=1 default=57343 value=70
                            hue (int)    : min=-127 max=127 step=1 default=-8193 value=0
                          gamma (int)    : min=1 max=8 step=1 default=57343 value=4
                           gain (int)    : min=0 max=65535 step=1 default=57343 value=16
           power_line_frequency (menu)   : min=0 max=2 default=1 value=1
                      sharpness (int)    : min=0 max=255 step=1 default=57343 value=3
```

Revanas, 2019-04-13:
@hawkeyexp did you have any success on creating the correct shader-UYVY.frag file? I have an easycap with stk1160 so I would be able to test.

hawkeyexp, 2019-05-29:
sorry for late response - have not really tracked it and started a new job on feb which consumed most time...
i can't really test it cause i have no stk1160 currently or a webcam using same profile.

meekys, 2019-06-08:
Hi @hawkeyexp,
I've added this fix into a new branch (cpp) after a 'bit' of refactoring
https://github.com/meekys/cam_overlay/tree/cpp
Let me know how this goes, if you get a chance to.

meekys, 2019-06-08:
@Revanas UYVY as opposed to YUVY ? (Yet another variation of the same thing by the looks of it?)

I've added another (untested) shader to support this variation in the cpp branch. It would be much appreciated if you were able to test this just to confirm it's all good.

Cheers,
Meekys