meekys/cam_overlay #10: MJPEG support

meekys, 2019-03-02:
Hi @meekys ,

Could you please also add the shader for MJPG. My device only supports MJPG pixel format. I would also appreciate any guidance on how to create the shader. Thanks
```bash
pi@raspberrypi:~/cam_overlay $ ./cam_overlay.bin
Opened device '/dev/video0'
Dimensions: 640 x 480 Pixel Format: MJPG
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
Opened screen 0 @ 800 x 480
Cannot open 'shader-MJPG.frag': 2, No such file or directory
```
```shell
pi@raspberrypi:~/cam_overlay $ v4l2-ctl --all
Driver Info (not using libv4l2):
        Driver name   : uvcvideo
        Card type     : USB2.0 PC CAMERA
        Bus info      : usb-3f980000.usb-1.5
        Driver version: 4.14.79
        Capabilities  : 0x84200001
                Video Capture
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps   : 0x04200001
                Video Capture
                Streaming
                Extended Pix Format
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
        Width/Height      : 640/480
        Pixel Format      : 'MJPG'
        Field             : None
        Bytes per Line    : 0
        Size Image        : 614400
        Colorspace        : Default
        Transfer Function : Default
        YCbCr/HSV Encoding: Default
        Quantization      : Default
        Flags             :
Crop Capability Video Capture:
        Bounds      : Left 0, Top 0, Width 640, Height 480
        Default     : Left 0, Top 0, Width 640, Height 480
        Pixel Aspect: 1/1
Selection: crop_default, Left 0, Top 0, Width 640, Height 480
Selection: crop_bounds, Left 0, Top 0, Width 640, Height 480
Streaming Parameters Video Capture:
        Capabilities     : timeperframe
        Frames per second: 60.000 (60/1)
        Read buffers     : 0
                     brightness (int)    : min=0 max=255 step=1 default=-8193 value=128
                       contrast (int)    : min=0 max=255 step=1 default=57343 value=128
                     saturation (int)    : min=0 max=255 step=1 default=57343 value=70
                            hue (int)    : min=-127 max=127 step=1 default=-8193 value=0
                          gamma (int)    : min=1 max=8 step=1 default=57343 value=4
                           gain (int)    : min=0 max=65535 step=1 default=57343 value=16
           power_line_frequency (menu)   : min=0 max=2 default=1 value=1
                      sharpness (int)    : min=0 max=255 step=1 default=57343 value=3
```

_Originally posted by @andreimoraru in https://github.com/meekys/cam_overlay/issues/7#issuecomment-468045599_

meekys, 2019-03-02:
Hi @andreimoraru,
This will need further research, as I haven't looked into whether a shader program can decode JPEG images and don't have the necessary hardware to test this scenario.

A possible solutions, if an MJPG shader cannot be be:

On the Raspberry pi:
- Using the OMX libraries to decode the image in hardware
In X11
- Using libjpeg or similar to decode it in software

meekys, 2019-03-02:
Possible shader that could be integrated
https://github.com/negge/jpeg_gpu

meekys, 2019-06-08:
Hi @andreimoraru,
I've added JPEG/MJPG support into a new branch (cpp) after a 'bit' of refactoring
https://github.com/meekys/cam_overlay/tree/cpp
Let me know how this goes, as I've only got one specific bit of hardware to test with.

Bishamon1987, 2019-07-14:
I can test this for you, got 2 video grabbers. 1 is MJPG other is YUVY.
But can you help me out how to install a branch? I think I'm doing it wrong...

meekys, 2019-07-15:
@Bishamon1987 That'd be great it you could.
Firstly, you'll need to check out the code, which you've undoubtable done.
`git clone https://github.com/meekys/cam_overlay.git`
Then from within the cam_overlay directory, switch to the cpp branch
`git checkout cpp`

After this, check the README.md for instructions on the dependencies and how to build. They're slightly different to the master branch.

Let me know if you have any further questions.

Bishamon1987, 2019-07-16:
Ok I'm trying to get the branch cpp but i get this error.
`pi@pi:~/cam_overlay $ sudo git checkout cpp
Already on 'cpp'
Your branch is up to date with 'origin/cpp'.`

the cmake. && make part gives me pages of errors.
and running the ./bin/camoverlay.bin isnt possible because it doesnt exist
in the /home/pi/cam_overlay/ folder

As I said I'm a noob, I need descriptive steps :P

meekys, 2019-07-17:
Haha, sure
Looks like you're already on the cpp branch, so that's good
Firstly, by the looks of it, you've checked out the code as root (by using sudo)
If possible, I'd remove the directory and start again, but without using sudo, otherwise you might run into permission issues (You could also prefix all build commands with sudo, but it's not recommended)

Running the build commands one by one. (Note the space between cmake and the . is important)
`cmake .`
And then if that succeeds without any errors
`make`

If they spit out a lot of errors, maybe grab the top 50 lines of the output and paste them in here?
If it scrolls off the screen, perhaps pipe the results to a log file, such as
`cmake . 2>&1 cmake.log`
`make 2>&1 make.log`

Bishamon1987, 2019-07-17:
Here is the log file.
[cmake error log.txt](https://github.com/meekys/cam_overlay/files/3403231/cmake.error.log.txt)

Bishamon1987, 2019-07-18:
By using no sudo gives me a lot of permission issues. That's why I used
sudo.
I try again tomorrow

Bishamon1987, 2019-07-19:
Well i got a lot further now.

```
pi@pi:~ $ git clone https://github.com/meekys/cam_overlay.git
Cloning into 'cam_overlay'...
remote: Enumerating objects: 65, done.
remote: Counting objects: 100% (65/65), done.
remote: Compressing objects: 100% (47/47), done.
remote: Total 311 (delta 18), reused 44 (delta 12), pack-reused 246
Receiving objects: 100% (311/311), 186.43 KiB | 792.00 KiB/s, done.
Resolving deltas: 100% (81/81), done.

pi@pi:~ $ cd cam_overlay/

pi@pi:~/cam_overlay $ git checkout cpp
Branch 'cpp' set up to track remote branch 'cpp' from 'origin'.
Switched to a new branch 'cpp'

pi@pi:~/cam_overlay $ cmake .
-- Found OpenGL: /usr/lib/arm-linux-gnueabihf/libOpenGL.so  found components:  EGL
-- Found ZLIB: /usr/lib/arm-linux-gnueabihf/libz.so (found version "1.2.11")
-- Found PNG: /usr/lib/arm-linux-gnueabihf/libpng.so (found version "1.6.36")
-- Looking for XOpenDisplay in /usr/lib/arm-linux-gnueabihf/libX11.so;/usr/lib/arm-linux-gnueabihf/libXext.so
-- Looking for XOpenDisplay in /usr/lib/arm-linux-gnueabihf/libX11.so;/usr/lib/arm-linux-gnueabihf/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/arm-linux-gnueabihf/libX11.so
-- Found JPEG: /usr/lib/arm-linux-gnueabihf/libjpeg.so (found version "62")
display_projects DisplayBcm;DisplayX11
input_projects InputV4L2;InputFile
output_projects OutputJpeg;OutputShader;OutputFile
display_modules Bcm;X11
input_modules V4L2MMap;V4L2UserPtr;V4L2Read;File
output_modules Jpeg;Shader;File
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/cam_overlay

pi@pi:~/cam_overlay $ make
Scanning dependencies of target core
[  2%] Building CXX object src/Core/CMakeFiles/core.dir/src/Common.cpp.o
[  5%] Linking CXX static library libcore.a
[  5%] Built target core
Scanning dependencies of target gl
[  7%] Building CXX object src/Gl/CMakeFiles/gl.dir/src/Buffer.cpp.o
[ 10%] Building CXX object src/Gl/CMakeFiles/gl.dir/src/FrameBuffer.cpp.o
[ 12%] Building CXX object src/Gl/CMakeFiles/gl.dir/src/GlContext.cpp.o
[ 15%] Building CXX object src/Gl/CMakeFiles/gl.dir/src/Program.cpp.o
[ 17%] Building CXX object src/Gl/CMakeFiles/gl.dir/src/Shader.cpp.o
[ 20%] Building CXX object src/Gl/CMakeFiles/gl.dir/src/Texture.cpp.o
[ 23%] Linking CXX static library libgl.a
[ 23%] Built target gl
Scanning dependencies of target camoverlay
[ 25%] Building CXX object src/CamOverlay/CMakeFiles/camoverlay.dir/src/Config.cpp.o
[ 28%] Building CXX object src/CamOverlay/CMakeFiles/camoverlay.dir/src/GlContext.cpp.o
[ 30%] Building CXX object src/CamOverlay/CMakeFiles/camoverlay.dir/src/Image.cpp.o
[ 33%] Building CXX object src/CamOverlay/CMakeFiles/camoverlay.dir/src/Logger.cpp.o
[ 35%] Building CXX object src/CamOverlay/CMakeFiles/camoverlay.dir/src/Math3d.cpp.o
[ 38%] Building CXX object src/CamOverlay/CMakeFiles/camoverlay.dir/src/Overlay.cpp.o
[ 41%] Building CXX object src/CamOverlay/CMakeFiles/camoverlay.dir/src/PngTexture.cpp.o
[ 43%] Building CXX object src/CamOverlay/CMakeFiles/camoverlay.dir/src/ScaledImage.cpp.o
[ 46%] Linking CXX static library libcamoverlay.a
[... 34 lines cut ...]
Copying shader-YUYV.frag
[ 87%] Built target OutputShader
Scanning dependencies of target OutputFile
[ 89%] Building CXX object src/Output/File/CMakeFiles/OutputFile.dir/src/OutputFile.cpp.o
[ 92%] Linking CXX static library libOutputFile.a
[ 92%] Built target OutputFile
Scanning dependencies of target camoverlay.bin
[ 94%] Building CXX object src/App/CMakeFiles/camoverlay.bin.dir/src/App.cpp.o
[ 97%] Building CXX object src/App/CMakeFiles/camoverlay.bin.dir/src/Main.cpp.o
/home/pi/cam_overlay/src/App/src/Main.cpp:14:46: error: use of 'auto' in parameter declaration only available with -fconcepts [-Werror]
 std::shared_ptr<IDisplay> CheckDisplay(const auto& injector, std::shared_ptr<Config> config, std::string module)
                                              ^~~~
/home/pi/cam_overlay/src/App/src/Main.cpp:23:42: error: use of 'auto' in parameter declaration only available with -fconcepts [-Werror]
 std::shared_ptr<IInput> CheckInput(const auto& injector, std::shared_ptr<Config> config, std::string module)
                                          ^~~~
/home/pi/cam_overlay/src/App/src/Main.cpp:32:44: error: use of 'auto' in parameter declaration only available with -fconcepts [-Werror]
 std::shared_ptr<IOutput> CheckOutput(const auto& injector, std::shared_ptr<Config> config, std::string module)
                                            ^~~~
cc1plus: all warnings being treated as errors
make[2]: *** [src/App/CMakeFiles/camoverlay.bin.dir/build.make:76: src/App/CMakeFiles/camoverlay.bin.dir/src/Main.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:767: src/App/CMakeFiles/camoverlay.bin.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

This last part is the problem now...

meekys, 2019-07-22:
Interesting. What version of gcc you have installed?
`gcc --version`

does the following want to install a newer version?
```
sudo apt-get update
sudo apt-get install build-essentials
```

Bishamon1987, 2019-07-22:
I did it all
The build-essentials didn’t work, incorrect

pi@CRANKSHAFT-NG:~/cam_overlay $ cmake .
CMake Error at /usr/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find PNG (missing: PNG_LIBRARY PNG_PNG_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.7/Modules/FindPNG.cmake:146 (find_package_handle_standard_args)
  src/CamOverlay/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/pi/cam_overlay/CMakeFiles/CMakeOutput.log".
pi@CRANKSHAFT-NG:~/cam_overlay $ gcc --version
gcc (Raspbian 6.3.0-18+rpi1+deb9u1) 6.3.0 20170516
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

meekys, 2019-07-28:
@Bishamon1987 Whoops, guess I shouldn't rely on my memory for these things. `build-essential` (without the trailing s)
In the case of your latest cmake error, you're missing the libpng dependency (libpng-dev). Did you start again on a clean install?

Bishamon1987, 2019-07-28:
Aah ok, well it's on my crankshaft NG install which is fresh so yes.
I'll try the build essential today

Bishamon1987, 2019-07-28:
ok so i tried it on my Crankshaft pi and Raspian Stretch. 
I think on crankshaft I could manage to fix it, `cmake `works good. But crashes on the `make `at 100%, maybe because there already is a cam_overlay on this system?
This is the error with make on crankshaft NG build
`[100%] Linking CXX executable ../../bin/camoverlay.bin
/opt/vc/lib/libbrcmEGL.so: undefined reference to `glDiscardFramebufferEXT'
/opt/vc/lib/libbrcmEGL.so: undefined reference to `glPointSizePointerOES'
/opt/vc/lib/libbrcmEGL.so: undefined reference to `glxx_set_error_api'
/opt/vc/lib/libbrcmEGL.so: undefined reference to `glxx_client_BindFramebuffer'
/opt/vc/lib/libbrcmEGL.so: undefined reference to `glintAttribPointer'
/opt/vc/lib/libbrcmEGL.so: undefined reference to `glxx_client_state_free'
/opt/vc/lib/libbrcmEGL.so: undefined reference to `glxx_client_FramebufferRenderbuffer'
/opt/vc/lib/libbrcmEGL.so: undefined reference to `glxx_buffer_info_set'
/opt/vc/lib/libbrcmEGL.so: undefined reference to `glxx_set_error'
/opt/vc/lib/libbrcmEGL.so: undefined reference to `glxx_client_DeleteFramebuffers'
/opt/vc/lib/libbrcmEGL.so: undefined reference to `gl11_client_state_init'
/opt/vc/lib/libbrcmEGL.so: undefined reference to `glxx_client_GetFramebufferAttachmentParameteriv'
/opt/vc/lib/libbrcmEGL.so: undefined reference to `glxx_client_IsRenderbuffer'
/opt/vc/lib/libbrcmEGL.so: undefined reference to `glxx_client_GetRenderbufferParameteriv'
/opt/vc/lib/libbrcmEGL.so: undefined reference to `glxx_client_DeleteRenderbuffers'
/opt/vc/lib/libbrcmEGL.so: undefined reference to `glxx_client_FramebufferTexture2D'
/opt/vc/lib/libbrcmEGL.so: undefined reference to `glxx_client_BindRenderbuffer'
/opt/vc/lib/libbrcmEGL.so: undefined reference to `glxx_client_GenerateMipmap'
/opt/vc/lib/libbrcmEGL.so: undefined reference to `glxx_client_RenderbufferStorage'
/opt/vc/lib/libbrcmEGL.so: undefined reference to `glxx_client_GenFramebuffers'
/opt/vc/lib/libbrcmEGL.so: undefined reference to `glxx_client_GenRenderbuffers'
/opt/vc/lib/libbrcmEGL.so: undefined reference to `glxx_client_CheckFramebufferStatus'
/opt/vc/lib/libbrcmEGL.so: undefined reference to `gl20_client_state_init'
/opt/vc/lib/libbrcmEGL.so: undefined reference to `glxx_buffer_info_get'
/opt/vc/lib/libbrcmEGL.so: undefined reference to `glxx_client_IsFramebuffer'
collect2: error: ld returned 1 exit status
src/App/CMakeFiles/camoverlay.bin.dir/build.make:139: recipe for target 'bin/camoverlay.bin' failed
make[2]: *** [bin/camoverlay.bin] Error 1
CMakeFiles/Makefile2:761: recipe for target 'src/App/CMakeFiles/camoverlay.bin.dir/all' failed
make[1]: *** [src/App/CMakeFiles/camoverlay.bin.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2`

On Rasbian Stretch, I updated everything, have the libpng_dev, libjpeg-dev, build-essential and gcc upgraded. But still no luck, to get `cmake .` working.

Getting this error in cmake . on rasbian
```
[ 97%] Building CXX object src/App/CMakeFiles/camoverlay.bin.dir/src/Main.cpp.o
/home/pi/cam_overlay/src/App/src/Main.cpp:14:46: error: use of ‘auto’ in parameter declaration only available with -fconcepts [-Werror]
 std::shared_ptr<IDisplay> CheckDisplay(const auto& injector, std::shared_ptr<Config> config, std::string module)
                                              ^~~~
/home/pi/cam_overlay/src/App/src/Main.cpp:23:42: error: use of ‘auto’ in parameter declaration only available with -fconcepts [-Werror]
 std::shared_ptr<IInput> CheckInput(const auto& injector, std::shared_ptr<Config> config, std::string module)
                                          ^~~~
/home/pi/cam_overlay/src/App/src/Main.cpp:32:44: error: use of ‘auto’ in parameter declaration only available with -fconcepts [-Werror]
 std::shared_ptr<IOutput> CheckOutput(const auto& injector, std::shared_ptr<Config> config, std::string module)
                                            ^~~~
cc1plus: all warnings being treated as errors
make[2]: *** [src/App/CMakeFiles/camoverlay.bin.dir/build.make:76: src/App/CMakeFiles/camoverlay.bin.dir/src/Main.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:767: src/App/CMakeFiles/camoverlay.bin.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

meekys, 2019-07-29:
Wow, looks like you've tried a lot to get this working.

Firstly, on Raspian Stretch where it's failing at 100% on make, this issue looks like the fact I'm using a 'fancy' auto keyword and the compiler isn't 100% happy for some reason and emitting a warning, which are treated as errors (annoying, but good practice)
```
/home/pi/cam_overlay/src/App/src/Main.cpp:14:46: error: use of 'auto' in parameter declaration only available with -fconcepts [-Werror]
 std::shared_ptr<IDisplay> CheckDisplay(const auto& injector, std::shared_ptr<Config> config, std::string module)
                                              ^~~~
```
I'm a bit stumped as to why this is failing for you (generating a warning) but is working for me.
I've been testing with gcc 7.4 on my desktop (Ubuntu), but had 6.3.0-18 on raspbian and both worked for me.
Untested, but you might be able to try this. It should theoretically set the CXXFLAGS to include the 'fconcepts' flag that g++ keeps asking for, then if this succeeds, try `make` again afterwards.
```
CXXFLAGS=-fconcepts cmake .
make
```

For the crankshaft-NG build, where it's failing with the following
`[100%] Linking CXX executable ../../bin/camoverlay.bin /opt/vc/lib/libbrcmEGL.so: undefined reference to glDiscardFramebufferEXT'`
This link https://www.reddit.com/r/RetroPie/comments/9um0yl/help_libbrcmeglso/ seems to suggest that simply reinstalling the following package will fix the missing file.
`sudo apt-get install --reinstall libraspberrypi0`

Bishamon1987, 2019-07-29:
Hmmm, so I did the on the crankshaft-NG build
`sudo apt-get install --reinstall libraspberrypi0`
command but it didn't do much good. Same error continues to exist...

meekys, 2019-07-30:
Hmm, I wonder which version of Raspbian crankshaft is built on.
Does this return any results?
`find / -name libbrcmEGL.so 2>/dev/null`
If not, how about this?
`find / -name libEGL.so 2>/dev/null`

The unresolved references is definitely sounding like an environmental issue. I think I'll have to swap out my SD card and try it under crankshaft-ng myself, see if we can't get to the bottom of this.

Bishamon1987, 2019-07-30:
These are some results, the ones you asked and some extras

find / -name libbrcmEGL.so 2>/dev/null
/opt/vc/lib/libbrcmEGL.so
/usr/lib/arm-linux-gnueabihf/libbrcmEGL.so

find / -name libEGL.so 2>/dev/null
/opt/vc/lib/libEGL.so
/usr/lib/arm-linux-gnueabihf/libEGL.so

uname -a
Linux CRANKSHAFT-NG 4.19.60-v7+ #1247 SMP Thu Jul 25 14:41:19 BST 2019 armv7l GNU/Linux

meekys, 2019-08-04:
Ok, so just confirming.
You're running the latest Raspbian Stretch? (It looks like it from the `uname -a` results)

This'll get you the os version/release that's installed
`cat /etc/os-release`

> PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
> NAME="Raspbian GNU/Linux"
> VERSION_ID="9"
> VERSION="9 (stretch)"
> ID=raspbian
> ID_LIKE=debian
> HOME_URL="http://www.raspbian.org/"
> SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
> BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

This'll ensure everything is updated to the latest packages
`sudo apt-get update && sudo apt-get upgrade`

Could you also check the version of g++ installed?
`g++ --version`

> g++ (Raspbian 6.3.0-18+rpi1+deb9u1) 6.3.0 20170516

Finally, can I get you to confirm which opengl is being detected by CMake?
`cat CMakeCache.txt | grep OPENGL`

On a raspbian OS, it should be showing something like this.
> OPENGL_INCLUDE_DIR:PATH=/opt/vc/include
> OPENGL_egl_LIBRARY:FILEPATH=/opt/vc/lib/libbrcmEGL.so
> OPENGL_gl_LIBRARY:FILEPATH=/opt/vc/lib/libbrcmEGL.so

Bishamon1987, 2019-08-04:
Seems like its correct right?

```
pi@CRANKSHAFT-NG:~ $ uname -a
Linux CRANKSHAFT-NG 4.19.63-v7+ #1249 SMP Thu Aug 1 16:26:30 BST 2019 armv7l GNU/Linux

pi@CRANKSHAFT-NG:~ $ cat /etc/os-release
PRETTY_NAME=Welcome to Crankshaft CarOS (2019-06-08 / Build E2B068B)
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

pi@CRANKSHAFT-NG:~ $ sudo apt-get update && sudo apt-get upgrade
Get:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Hit:2 http://archive.raspberrypi.org/debian stretch InRelease
Fetched 15.0 kB in 1s (9752 B/s)
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

pi@CRANKSHAFT-NG:~ $ g++ --version
g++ (Raspbian 6.3.0-18+rpi1+deb9u1) 6.3.0 20170516
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

pi@CRANKSHAFT-NG:~ $ cd cam_overlay/
pi@CRANKSHAFT-NG:~/cam_overlay $ cat CMakeCache.txt | grep OPENGL
OPENGL_INCLUDE_DIR:PATH=/opt/vc/include
OPENGL_egl_LIBRARY:FILEPATH=/opt/vc/lib/libbrcmEGL.so
OPENGL_gl_LIBRARY:FILEPATH=/opt/vc/lib/libbrcmEGL.so
OPENGL_glu_LIBRARY:FILEPATH=/usr/lib/arm-linux-gnueabihf/libGLU.so
OPENGL_xmesa_INCLUDE_DIR:PATH=OPENGL_xmesa_INCLUDE_DIR-NOTFOUND
//ADVANCED property for variable: OPENGL_INCLUDE_DIR
OPENGL_INCLUDE_DIR-ADVANCED:INTERNAL=1
//ADVANCED property for variable: OPENGL_gl_LIBRARY
OPENGL_gl_LIBRARY-ADVANCED:INTERNAL=1
//ADVANCED property for variable: OPENGL_glu_LIBRARY
OPENGL_glu_LIBRARY-ADVANCED:INTERNAL=1
//ADVANCED property for variable: OPENGL_xmesa_INCLUDE_DIR
OPENGL_xmesa_INCLUDE_DIR-ADVANCED:INTERNAL=1
```

meekys, 2019-08-04:
Yeah, that looks about what I'd expect
If you reset your build environment and re-run cmake
ie.

```
rm CMakeCache.txt
rm -rf CMakeFiles
cmake .
```

For me, this is what I get
```
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenGL: /opt/vc/lib/libbrcmEGL.so
-- Found ZLIB: /usr/lib/arm-linux-gnueabihf/libz.so (found version "1.2.8")
-- Found PNG: /usr/lib/arm-linux-gnueabihf/libpng.so (found version "1.6.28")
-- Looking for XOpenDisplay in /usr/lib/arm-linux-gnueabihf/libX11.so;/usr/lib/arm-linux-gnueabihf/libXext.so
-- Looking for XOpenDisplay in /usr/lib/arm-linux-gnueabihf/libX11.so;/usr/lib/arm-linux-gnueabihf/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Found X11: /usr/lib/arm-linux-gnueabihf/libX11.so
-- Found JPEG: /usr/lib/arm-linux-gnueabihf/libjpeg.so
display_projects DisplayBcm;DisplayX11
input_projects InputV4L2;InputFile
output_projects OutputJpeg;OutputShader;OutputFile
display_modules Bcm;X11
input_modules V4L2MMap;V4L2UserPtr;V4L2Read;File
output_modules Jpeg;Shader;File
-- Configuring done
-- Generating done
```

Bishamon1987, 2019-08-04:
```pi@CRANKSHAFT-NG:~ $ cd cam_overlay/
pi@CRANKSHAFT-NG:~/cam_overlay $ rm CMakeCache.txt
pi@CRANKSHAFT-NG:~/cam_overlay $ rm -rf CMakeFiles
pi@CRANKSHAFT-NG:~/cam_overlay $ cmake .
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenGL: /opt/vc/lib/libbrcmEGL.so
-- Found ZLIB: /usr/lib/arm-linux-gnueabihf/libz.so (found version "1.2.8")
-- Found PNG: /usr/lib/arm-linux-gnueabihf/libpng.so (found version "1.6.28")
-- Looking for XOpenDisplay in /usr/lib/arm-linux-gnueabihf/libX11.so;/usr/lib/arm-linux-gnueabihf/libXext.so
-- Looking for XOpenDisplay in /usr/lib/arm-linux-gnueabihf/libX11.so;/usr/lib/arm-linux-gnueabihf/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/arm-linux-gnueabihf/libX11.so
-- Found JPEG: /usr/lib/arm-linux-gnueabihf/libjpeg.so
display_projects DisplayBcm;DisplayX11
input_projects InputV4L2;InputFile
output_projects OutputJpeg;OutputShader;OutputFile
display_modules Bcm;X11
input_modules V4L2MMap;V4L2UserPtr;V4L2Read;File
output_modules Jpeg;Shader;File
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/cam_overlay
```

Bishamon1987, 2019-08-06:
DUDE IT WORKS !!
It's pretty grainy though.
Look at this image
https://photos.app.goo.gl/JysLzNstATjCuFCr8
and gives these results
` -s
Initialising DisplayBcm...
Initialising GL context...
Initialising InputV4L2MMap...
Opened device '/dev/video1'
Initialising OutputJpeg...
Dimensions: 640 x 480 Pixel Format: MJPG
Running...
Premature end of JPEG file
Premature end of JPEG file
Premature end of JPEG file`

Every Premature end of JPEG file gives a short (something like 0,1 second) black screen

meekys, 2019-08-08:
Wow, glad we finally got it building!!! What was required in the end?

I'd say, chances are the graininess is due to compression artefacts in the JPEG images themselves (the web cam might compress them a bit too much). Currently no post-processing is done on the image to 'smooth' it out.

The premature end of JPEG file seems like a bug... probably the buffer not big enough.
I can get a similar error (very intermittently) of `Corrupt JPEG data: premature end of data segment` which may or may not be related. I'll have a look into fixing that if I can, which might fix your issue too.

closed 2020-05-18 by meekys