meekys/cam_overlay #12: * failed to add service - already in use? when using dtoverlay=vc4-fkms-v3d

rubenverhoef, 2019-03-27:
I get :
` * failed to add service - already in use?`
when i'm using `dtoverlay=vc4-fkms-v3d` in my config.txt on my RPI
Any ideas?

meekys, 2019-03-28:
Hi @rubenverhoef ,

This looks like this might not be compatible with the display method that cam_overlay uses, dispmanx overlay and gles

https://www.raspberrypi.org/forums/viewtopic.php?t=159853
> The standard camera and video decode apps that drive the display from the gpu cannot be used when vc4-kms-v3d is enabled.
Apps that use dispmanx for overlays or performance reasons (e.g. emulators) cannot be used.

If you're using X11, you could try compiling it in that mode?
`make clean && make DISPLAY=x11`

rubenverhoef, 2019-03-28:
Okay i will try and see what the performance and usability are.

closed 2019-04-30 by rubenverhoef