meekys/cam_overlay #13: VIDIOC_REQBUFS error 12 when firing cam_overlay

enriquedelpino, 2019-03-27:
Hi there,

I'm just trying to get started using cam_overlay (through crankshaft) to fire up a rear cam in my car. I'm getting a memory issue with the webcam which I'm using to test my setup, and I am not sure whether the problem is coming from my setup, or rather if this is a problem someone has come across before:

`pi@CRANKSHAFT-NG:/opt/crankshaft/cam_overlay $ ./cam_overlay.bin 
Opened device '/dev/video0'
Dimensions: 640 x 480 Pixel Format: YUYV
VIDIOC_REQBUFS error 12, Cannot allocate memory`

Also I can provide the details of the camera used:

`pi@CRANKSHAFT-NG:/opt/crankshaft/cam_overlay $ v4l2-ctl --all
Driver Info (not using libv4l2):
	Driver name   : uvcvideo
	Card type     : Microsoft® LifeCam HD-3000: Mi
	Bus info      : usb-3f980000.usb-1.3
	Driver version: 4.14.98
	Capabilities  : 0x84200001
		Video Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps   : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height      : 640/480
	Pixel Format      : 'YUYV'
	Field             : None
	Bytes per Line    : 1280
	Size Image        : 3085365528
	Colorspace        : sRGB
	Transfer Function : Default
	YCbCr/HSV Encoding: Default
	Quantization      : Default
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 640, Height 480
	Default     : Left 0, Top 0, Width 640, Height 480
	Pixel Aspect: 1/1
Selection: crop_default, Left 0, Top 0, Width 640, Height 480
Selection: crop_bounds, Left 0, Top 0, Width 640, Height 480
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 0.003 (1/353)
	Read buffers     : 0
                     brightness (int)    : min=30 max=255 step=1 default=-8193 value=142
                       contrast (int)    : min=0 max=10 step=1 default=57343 value=5
                     saturation (int)    : min=0 max=200 step=1 default=57343 value=83
 white_balance_temperature_auto (bool)   : default=1 value=1
           power_line_frequency (menu)   : min=0 max=2 default=2 value=0
      white_balance_temperature (int)    : min=2800 max=10000 step=1 default=57343 value=4500 flags=inactive
                      sharpness (int)    : min=0 max=50 step=1 default=57343 value=25
         backlight_compensation (int)    : min=0 max=10 step=1 default=57343 value=5
                  exposure_auto (menu)   : min=0 max=3 default=0 value=3
              exposure_absolute (int)    : min=5 max=20000 step=1 default=156 value=156 flags=inactive
                   pan_absolute (int)    : min=-201600 max=201600 step=3600 default=0 value=0
                  tilt_absolute (int)    : min=-201600 max=201600 step=3600 default=0 value=0
                  zoom_absolute (int)    : min=0 max=10 step=1 default=57343 value=0`

meekys, 2019-03-28:
Hi @enriquedelpino,

Thanks for the details. I'm currently in the middle of moving, so I can't look into this into too much depth right now, but a few things to try.

Could you try adding --userp as a command line option to see whether this works?
`./cam_overlay.bin --userp`

Does the device work when using v4l2-ctl directly?
`v4l2-ctl --stream-mmap --stream-count=1 -d /dev/video0 --stream-to=output.dat --verbose`

enriquedelpino, 2019-03-28:
Thanks for the quick response @meekys,

This is the output from the commands you've suggested to run:

```
pi@CRANKSHAFT-NG:/opt/crankshaft/cam_overlay $ ./cam_overlay.bin --userp
Opened device '/dev/video0'
Dimensions: 640 x 480 Pixel Format: YUYV
Out of memory
```
```
pi@CRANKSHAFT-NG:/tmp $ v4l2-ctl --stream-mmap --stream-count=1 -d /dev/video0 --stream-to=output.dat --verbose
VIDIOC_QUERYCAP: ok
VIDIOC_REQBUFS: failed: Cannot allocate memory
```

There is something extra I did not mention earlier on. I have been able to have the camera getting a screenshot using `fswebcam` and also to record a short video using `ffmpeg`, so it seems the camera runs happily on the RPi

```
pi@CRANKSHAFT-NG:/tmp $ ffmpeg -t 120 -f v4l2 -framerate 25 -video_size 640x80 -i /dev/video0 output.mkv
```

meekys, 2019-05-27:
@enriquedelpino Sorry I've taken a while to get back to you on this. Been caught up with life and a re-write.
I have to admit, it looks like it's a memory issue.
I wonder if it's related to this line
Size Image        : 3085365528
If so, the poor system might be trying to allocate ~2942.4MB
It's interesting that the v4l2 input from ffmpeg doesn't seem to be having issues with this, I wonder what they do differently.

meekys, 2019-05-27:
@enriquedelpino Too add to this, could you try the following if you're able to compile from source?
https://github.com/meekys/cam_overlay/blob/master/cam_overlay.c#L905
Change `fmt.fmt.pix.sizeimage` to `fmt.fmt.pix.bytesperline * fmt.fmt.pix.height` and see if this resolves the issue?

meekys, 2019-06-08:
Hi @enriquedelpino,
I've added this fix into a new branch (cpp) after a 'bit' of refactoring
https://github.com/meekys/cam_overlay/tree/cpp
Let me know how this goes.

Bishamon1987, 2019-08-08:
I used the cpp branch
It somehow gives this error with one of my videograbbers but doesn't give the error with the master branch. (Version used in Crankshaft NG alpha 5)

meekys, 2019-08-11:
@Bishamon1987 Yeah, it's an odd error. From what I can tell the driver was reporting a stupidly big sizeimage vs what was required, resulting in an out of memory error. (If it's the same issue)
That being said, I thought/hoped it might have been fixed on the cpp branch.

Could I get the output of `v4l2-ctl --all` when both devices are plugged in. We might be able to figure out what they're reporting and figure out why this is occurring for one device, but not the other.