meekys/cam_overlay #14: [RPI][X11] Compile failure

rubenverhoef, 2019-03-28:
When compiling on my RPI 3b+ with X11 mode i get this output:

````
pi@raspberrypi:~/cam_overlay $ make DISPLAY=x11
cc -Ix11  -DSTANDALONE -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DTARGET_POSIX -D_LINUX -fPIC -DPIC -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -U_FORTIFY_SOURCE -Wall -g -ftree-vectorize -pipe -Wno-psabi  -g -c cam_overlay.c -o cam_overlay.o -Wno-deprecated-declarations
In file included from x11/gl_common.h:2:0,
                 from cam_overlay.c:32:
/usr/include/GLES2/gl2.h:73:25: error: conflicting types for ‘GLsizeiptr’
 typedef khronos_ssize_t GLsizeiptr;
                         ^~~~~~~~~~
In file included from /usr/include/GL/gl.h:2055:0,
                 from x11/gl_common.h:1,
                 from cam_overlay.c:32:
/usr/include/GL/glext.h:468:19: note: previous declaration of ‘GLsizeiptr’ was here
 typedef ptrdiff_t GLsizeiptr;
                   ^~~~~~~~~~
In file included from x11/gl_common.h:2:0,
                 from cam_overlay.c:32:
/usr/include/GLES2/gl2.h:74:26: error: conflicting types for ‘GLintptr’
 typedef khronos_intptr_t GLintptr;
                          ^~~~~~~~
In file included from /usr/include/GL/gl.h:2055:0,
                 from x11/gl_common.h:1,
                 from cam_overlay.c:32:
/usr/include/GL/glext.h:469:19: note: previous declaration of ‘GLintptr’ was here
 typedef ptrdiff_t GLintptr;
                   ^~~~~~~~
In file included from x11/gl_common.h:2:0,
                 from cam_overlay.c:32:
/usr/include/GLES2/gl2.h:393:28: error: conflicting types for ‘PFNGLBUFFERDATAPROC’
 typedef void (GL_APIENTRYP PFNGLBUFFERDATAPROC) (GLenum target, GLsizeiptr size, const void *data, GLenum usage);
                            ^~~~~~~~~~~~~~~~~~~
In file included from /usr/include/GL/gl.h:2055:0,
                 from x11/gl_common.h:1,
                 from cam_overlay.c:32:
/usr/include/GL/glext.h:532:25: note: previous declaration of ‘PFNGLBUFFERDATAPROC’ was here
 typedef void (APIENTRYP PFNGLBUFFERDATAPROC) (GLenum target, GLsizeiptr size, const void *data, GLenum usage);
                         ^~~~~~~~~~~~~~~~~~~
In file included from x11/gl_common.h:2:0,
                 from cam_overlay.c:32:
/usr/include/GLES2/gl2.h:394:28: error: conflicting types for ‘PFNGLBUFFERSUBDATAPROC’
 typedef void (GL_APIENTRYP PFNGLBUFFERSUBDATAPROC) (GLenum target, GLintptr offset, GLsizeiptr size, const void *data);
                            ^~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/GL/gl.h:2055:0,
                 from x11/gl_common.h:1,
                 from cam_overlay.c:32:
/usr/include/GL/glext.h:533:25: note: previous declaration of ‘PFNGLBUFFERSUBDATAPROC’ was here
 typedef void (APIENTRYP PFNGLBUFFERSUBDATAPROC) (GLenum target, GLintptr offset, GLsizeiptr size, const void *data);
                         ^~~~~~~~~~~~~~~~~~~~~~
Makefile.include:14: recipe for target 'cam_overlay.o' failed
make: *** [cam_overlay.o] Error 1
````

When removing the line `#include "GL/gl.h"` from `x11/gl_common.h` i will get:
````
pi@raspberrypi:~/cam_overlay $ make DISPLAY=x11
cc -Ix11  -DSTANDALONE -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DTARGET_POSIX -D_LINUX -fPIC -DPIC -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -U_FORTIFY_SOURCE -Wall -g -ftree-vectorize -pipe -Wno-psabi  -g -c cam_overlay.c -o cam_overlay.o -Wno-deprecated-declarations
cc -Ix11  -DSTANDALONE -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DTARGET_POSIX -D_LINUX -fPIC -DPIC -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -U_FORTIFY_SOURCE -Wall -g -ftree-vectorize -pipe -Wno-psabi  -g -c common.c -o common.o -Wno-deprecated-declarations
cc -Ix11  -DSTANDALONE -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DTARGET_POSIX -D_LINUX -fPIC -DPIC -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -U_FORTIFY_SOURCE -Wall -g -ftree-vectorize -pipe -Wno-psabi  -g -c display.c -o display.o -Wno-deprecated-declarations
cc -Ix11  -DSTANDALONE -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DTARGET_POSIX -D_LINUX -fPIC -DPIC -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -U_FORTIFY_SOURCE -Wall -g -ftree-vectorize -pipe -Wno-psabi  -g -c x11/display.c -o x11/display.o -Wno-deprecated-declarations
cc -Ix11  -DSTANDALONE -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DTARGET_POSIX -D_LINUX -fPIC -DPIC -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -U_FORTIFY_SOURCE -Wall -g -ftree-vectorize -pipe -Wno-psabi  -g -c png_texture.c -o png_texture.o -Wno-deprecated-declarations
cc -o cam_overlay.bin -Wl,--whole-archive cam_overlay.o common.o display.o x11/display.o png_texture.o -lpng std=c99 -lm -lEGL -lGLESv2 -lX11 -Wl,--no-whole-archive -rdynamic
cc: error: std=c99: No such file or directory
Makefile.include:22: recipe for target 'cam_overlay.bin' failed
make: *** [cam_overlay.bin] Error 1
rm cam_overlay.o common.o png_texture.o display.o x11/display.o
````

any ideas?

meekys, 2019-04-28:
@rubenverhoef Sorry for the slow reply, life's been a 'bit' busy the last few weeks.
It looks like there might be a bug in the Makefile for x11

If you edit `x11/Makefile.include` and add a - before std=c99 this might fix the issue you're having.
ie.
`LDFLAGS+=-std=c99 -lm -lEGL -lGLESv2 -lX11`

rubenverhoef, 2019-04-30:
@meekys No problem we all have normal lifes to live

Stupid that i not saw it, i was misled by the second **=** i think.
The first fix is also needed for me to compile the software. It is working fine after that.
So close #12 

I also added some changes to run it fullscreen on my pi, like the not X11 version does: https://github.com/rubenverhoef/cam_overlay/commit/06f2f112747177551ba62e54190424592b5b91b4

closed 2019-04-30 by rubenverhoef