meekys/cam_overlay #15: split screen front and rearview

Bishamon1987, 2019-05-18:
I was wondering if it is possible to add a second camera view. And a way to switch back and forth.
Something like the WhatsApp video call view.
One little screen for one of the views and a big view window for the other cam
Or just split screen.
I would like a rear and front camera for easier parking.
But I also want to use them as dashcams.
The front camera will be mounted underneath the front licenseplate

meekys, 2019-05-27:
Hi @Bishamon1987 
This should theoretically be possible by running multiple instances of cam_overlay
Currently cam_overlay is fixed to one display layer and will attempt to scale the image to full screen, but the code could be tweaked to position the displayed image to whatever position you wanted.

Bishamon1987, 2019-08-06:
Can you tell me where in the code i can change the display size and position?

meekys, 2019-08-08:
In the cpp branch, have a look at the ScaledImage class.
This is what currently transforms the image. It does things like stretching, rotating or flipping and renders the resulting image to the screen

Bishamon1987, 2019-08-09:
so I tried it but its really buggy. It even crashed my open auto which doesnt function anymore in X11 for some reason