meekys/cam_overlay #16: Possible to add option without guidelines

Bishamon1987, 2019-07-14:
Hey,
I have Crankshaft NG running on my pi 3B+.
I want to add a front and back camera. 
I will use a switch to change the views. 
Pressing forward is front cam. Middle stand is camera off. Pulling backwards is rearcam.
I dont need any guidlines for the front view. But i need them for the back. 
Is it possible to make a option for this?
Thanks

meekys, 2019-07-15:
@Bishamon1987 Sure, that should be possible as a new feature.
For now, a workaround would be simply to edit the overlay.png and make it all transparent.

Bishamon1987, 2019-07-17:
Maybe an even better solution would be this:
option for which guideline.png to use.

I have two camera's, front and rear. So I need different images for each. Since the height and vertical angles are different.

meekys, 2019-08-22:
Added option to specify an overlay file. -O or --overlay, with a default of res/Overlay.png
Must be a relative path to cam_overlay.bin

Bishamon1987, 2019-08-22:
That's great! Now I have to directories with camoverlay.
One called rearcam and other called frontcam

closed 2020-05-18 by meekys