meekys/cam_overlay #18: overlay.png not rendered correct

Bishamon1987, 2019-08-08:
Im not sure of my title is correct.

I made a overlay.png that's should be used full screen since my car dimensions are very wide my lines start from the sides instead of the bottom. 
It looks like cam overlay change the picture in dimensions. But I need it rendered full screen like the cam input itself.

Bishamon1987, 2019-08-08:
What it shows when turning on camera...
[image]

What it shows without camera feed...
[image]

### **How I want to show on screen** (the guidelines are removable on the camera)...
[image]

Bishamon1987, 2019-08-09:
Ok, so I found my solution.
```
cd ~
git clone https://github.com/meekys/cam_overlay.git
cd ~/cam_overlay 
git checkout cpp
cd cam_overlay/src/CamOverlay/src/
sudo nano Overlay.cpp
```

**Then change this part**
```
VERTEX_T Overlay::ImageQuad[4] =
{   // x,     y,     z,     u,    v
    {{-1.0f,  1.0f,  2.0f}, 0.0f, 1.0f}, // Top left
    {{-1.0f, -1.0f,  1.0f}, 0.0f, 0.0f}, // Bottom left
    {{ 1.0f,  1.0f,  2.0f}, 1.0f, 1.0f}, // Top right
    {{ 1.0f, -1.0f,  1.0f}, 1.0f, 0.0f}  // Bottom right
};
```

**To this** (the Z values should all be 1.0f)
```
VERTEX_T Overlay::ImageQuad[4] =
{   // x,     y,     z,     u,    v
    {{-1.0f,  1.0f,  1.0f}, 0.0f, 1.0f}, // Top left
    {{-1.0f, -1.0f,  1.0f}, 0.0f, 0.0f}, // Bottom left
    {{ 1.0f,  1.0f,  1.0f}, 1.0f, 1.0f}, // Top right
    {{ 1.0f, -1.0f,  1.0f}, 1.0f, 0.0f}  // Bottom right
};
```

The Z values give perspective to the rendered lines which are horizontal and vertical and give it depth.

meekys, 2019-08-11:
@Bishamon1987 Yeah, you're correct. I force a perspective. I should probably put this into an option to allow people to opt-in or opt-out depending on their preferences.
The semi-transparency looks awesome in that overlay btw :-)

Bishamon1987, 2019-08-12:
VERTEX_T Overlay::ImageQuad[4] =
{   // x,     y,     z,     u,    v
    {{-1.0f,  1.0f,  1.0f}, 0.0f, 1.0f}, // Top left
    {{-1.0f, -1.0f,  1.0f}, 0.0f, 0.0f}, // Bottom left
    {{ 1.0f,  1.0f,  1.0f}, 1.0f, 1.0f}, // Top right
    {{ 1.0f, -1.0f,  1.0f}, 1.0f, 0.0f}  // Bottom right
};

Btw just a question, what are these u and v values? are they the x and y of the videofeed?