meekys/cam_overlay #19: in cpp branch -u and --userp dont work

Bishamon1987, 2019-08-09:
~/cam_overlay-full/bin $ ./camoverlay.bin -d /dev/video0 --userp
./camoverlay.bin: unrecognized option '--userp'
Usage: ./camoverlay.bin [options]

Version 1.3

AND

~/cam_overlay-full/bin $ ./camoverlay.bin -d /dev/video0 -u
./camoverlay.bin: invalid option -- 'u'
Usage: ./camoverlay.bin [options]

Version 1.3
Options:
-? | --help          Print this message
-d | --device name   Video device name [/dev/video0]
-i | --input         Input module [V4L2MMap]
-o | --output        Output module [auto]
-D | --display       Display module [X11]
-s | --stretch       Stretch image to screen
-r | --rotate        Rotate image 180 degrees
-h | --fliph         Flip image horizontally
-v | --flipv         Flip image vertically

The -u and --userp are writen in the Readme but i dont see it in the options as you see above.

meekys, 2019-08-11:
Yeah, that looks like a bug in the readme. That option no longer exists using --userp
If you wan't to use that feature now, try `--input V4L2UserPtr` instead

meekys, 2019-08-22:
This should now be removed from the readme file

closed 2020-05-18 by meekys