meekys/cam_overlay #20: DisplayBCM error in CPP branch using RCA output

Bishamon1987, 2019-08-14:
I got some new issue, I'm not sure whats the problem yet.

When I'm running the camoverlay.bin I get this error
`Initialising DisplayBcm...
rearcam.bin: /home/pi/cam_overlay/src/Display/Bcm/src/DisplayBcm.cpp:16: DisplayBcm::DisplayBcm(std::shared_ptr<ILogger>): Assertion `success >= 0' failed.
Aborted`

The weird thing is when the pi is connected via HDMI to my television it works fine. 
But in the car I have to use a RCA connection. (I don't want to use more energy by using these HDMI converters, I have a PHEV)

While I'm typing it I think might this be caused by my boot/config.txt where my HDMI is not ingored?

meekys, 2019-08-16:
Looks like this assertion is happening because it's failing to get the display size.
The display is hard coded to 0, which I think is correct.
It might be worth trying to disable hdmi in the boot/config.txt so that you're certain it uses RCA only?

Bishamon1987, 2019-08-16:
I already tried that, no luck.
I think it might be an issue caused by Crankshaft-NG. Because it did work before I updated it.
I reported the issue to the Crankshaft developers

Bishamon1987, 2019-08-16:
Is there a way to see which display number the minijack port is using?

meekys, 2019-08-22:
Honestly not sure. All the docs/examples just have a hard coded value of 0.
I'd assuming that the hdmi input was disabled (ie, not detected on startup) that it would be the primary display (ie, 0)

Not sure whether this'll give us any insight... but what does this output?
`sudo vcdbg log msg`

meekys, 2019-08-22:
I've also tweaked the error message being displayed so it might be a bit more useful (or not)
It'll show the location of the error and the return code.

Bishamon1987, 2019-08-22:
It seems to be an issue with crankshaft. It want to run in HDMI instead of taking the available display type

meekys, 2019-08-29:
Probably something that can be tweaked in the boot/config.txt ?