meekys/cam_overlay #22: dependencies to run cam_overlay.bin

Bishamon1987, 2019-08-21:
What should be loaded to get the cam_overlay.bin runnning?
I want it to show the rearcam before crankshaft is done with loading.

meekys, 2019-08-22:
It should be able to run completely standalone, unless you're using the X11 display (on the cpp branch)

Bishamon1987, 2019-08-22:
Well yes I do use x11 indeed.

meekys, 2019-08-22:
Nah, I mean if the display mode for cam_overlay.bin is set to X11
The display should default to Bcm if that's available, which means it'll render directly to the screen, so no X11 is required.

Bishamon1987, 2019-08-22:
I didn't set -D X11 so I think it runs in bmc?
Also the mpeg mode doesnt show great colors. It more grayish then full color. So maybe the shader is off?

meekys, 2019-08-29:
-D X11 should only be required on the master branch.
In the cpp branch, it'll auto-detect dependencies, so both Bcm and X11 would be available

When using the Bcm display (default for a raspberry pi), it'll work without X11.
When using the X11 display, you'll see a window border around the output, obviously it'll only work under X11 in this mode (Mostly so I can test w/out doing all the dev work on my Pi)

closed 2020-05-25 by meekys