meekys/cam_overlay #23: Stretch doesn't work with video input mjpeg

Bishamon1987, 2020-05-23:
Hello, 
I got a camera attached to a video grabber with the following parameters.: 
`Video: mjpeg, yuvj422p(pc, bt470bg/unknown/unknown), 640x480, -5 kb/s, 30 fps, 30 tbr, 1000k tbn, 1000k tbc`

For some reason camoverlay wont use the stretch option with this video input.
The picture down here shows only half the video input (only right side). 
Also the colors are really off...
[image]

What do you think is the problem? How can I fix it?

meekys, 2020-05-25:
Is this from the cpp branch?
If so, what's the output when you run this?
It's possible the colours are off, as the JPEG shader I've implemented is pretty basic and looks to only really handle a single encoding type.

If you could run `cam-overlay --output File` it should output a txt and dat file with the current date/time. This would help a lot with testing this.

Bishamon1987, 2020-05-25:
It is the cpp branch but not updated in the last 4-6 months.
When i have time i will make the file :)

Bishamon1987, 2020-06-20:
Well so finally i had the time to get the files you requested ^^
[txt and dat files.zip](https://github.com/meekys/cam_overlay/files/4808126/txt.and.dat.files.zip)