meekys/cam_overlay #25: Can't get it work on RPI4 buster/crankshaft, help

egisz, 2020-07-29:
Hi,
I'm struggling to connect RCA camera, connected through USB receiver, connected to RPI4 Buster/Crankshaft, @fazerxlo [branch ](https://github.com/fazerxlo/crankshaft.git)
I use EasyCap receiver, like [this ](https://user-images.githubusercontent.com/40425388/60817626-51bfb880-a19c-11e9-9ecf-8b2ba122704f.jpeg)
Rear camera is [this one](https://www.aliexpress.com/item/32918652253.html?spm=a2g0s.9042311.0.0.294a4c4dXx20ez), 

Tried to build cam_overlay, both master and cpp branch, but no luck, 
Tried to enable/disable rpi-config fake KMS. No luck.
master branch gives error:
```
pi@raspberrypi:~/cam_overlay $ ./cam_overlay.bin
Opened device '/dev/video0'
Dimensions: 640 x 480 Pixel Format: MJPG
cam_overlay.bin: cam_overlay.c:105: init_ogl: Assertion `EGL_FALSE != result' failed.
Aborted
```
with cpp branch had issues building, in the end it also does not work:
```
pi@raspberrypi:~ $ ./cam_overlay_cpp/cam_overlay/bin/camoverlay.bin
Initialising DisplayBcm...
Initialising GL context...
* failed to add service - already in use?
```
Please help.

egisz, 2020-07-30:
I guess the problem ischipset of Easycap stick: 
```
#pi@raspberrypi:/home $ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 009: ID 18ec:5555 Arkmicro Technologies Inc.
```
It appears in dmesg output, with some warnings:
```
[    5.008212] vc4-drm gpu: fb0: vc4drmfb frame buffer device
[    5.012612] usb 1-1.3.4: New USB device found, idVendor=18ec, idProduct=5555, bcdDevice= 1.00
[    5.012632] usb 1-1.3.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    5.012648] usb 1-1.3.4: Product: USB2.0 PC CAMERA
[    5.012664] usb 1-1.3.4: Manufacturer: ARKMICRO
[    5.030425] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    5.035233] usb 1-1.3.4: 3:1: cannot get freq at ep 0x82
[    5.093315] usb 1-1.3.4: Warning! Unlikely big volume range (=11903), cval->res is probably wrong.
[    5.093337] usb 1-1.3.4: [2] FU [Mic Capture Volume] ch = 2, val = 384/12287/1
[    5.095481] brcmfmac: F1 signature read @0x18000000=0x15264345
[    5.112253] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.112828] usbcore: registered new interface driver brcmfmac
[    5.138988] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt failed with error -2
[    5.149002] rpivid-mem feb00000.hevc-decoder: rpivid-hevcmem initialised: Registers at 0xfeb00000 length 0x00010000
[    5.149437] rpivid-mem feb10000.rpivid-local-intc: rpivid-intcmem initialised: Registers at 0xfeb10000 length 0x00001000
[    5.149851] rpivid-mem feb20000.h264-decoder: rpivid-h264mem initialised: Registers at 0xfeb20000 length 0x00010000
[    5.150250] rpivid-mem feb30000.vp9-decoder: rpivid-vp9mem initialised: Registers at 0xfeb30000 length 0x00010000
[    5.376445] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.398580] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  2 2020 23:30:41 version 7.45.202 (r724630 CY) FWID 01-72f6ece2
[    5.944894] uvcvideo: Found UVC 1.00 device USB2.0 PC CAMERA (18ec:5555)
[    5.983078] uvcvideo: UVC non compliance - GET_DEF(PROBE) not supported. Enabling workaround.
[    6.006746] uvcvideo 1-1.3.4:1.0: Entity type for entity Processing 2 was not initialized!
[    6.006770] uvcvideo 1-1.3.4:1.0: Entity type for entity Camera 1 was not initialized!
[    6.010790] input: USB2.0 PC CAMERA as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3.4/1-1.3.4:1.0/input/input7
[    6.018993] usbcore: registered new interface driver uvcvideo
[    6.019010] USB Video Class driver (1.1.1)

```

egisz, 2020-07-30:
I succeeded to open video using mplayer:
```
pi@raspberrypi:~ $ mplayer tv:// -tv device=/dev/video0 
MPlayer 1.3.0 (Debian), built with gcc-8 (C) 2000-2016 MPlayer Team
do_connect: could not connect to socket
connect: No such file or directory
Failed to open LIRC support. You will not be able to use your remote control.

Playing tv://.
TV file format detected.
Selected driver: v4l2
 name: Video 4 Linux 2 input
 author: Martin Olschewski <olschewski@zpr.uni-koeln.de>
 comment: first try, more to come ;-)
v4l2: your device driver does not support VIDIOC_G_STD ioctl, VIDIOC_G_PARM was used instead.
Selected device: USB2.0 PC CAMERA
 Capabilities:  video capture  streaming
 supported norms:
 inputs: 0 = Camera 1;
 Current input: 0
 Current format: MJPEG
tv.c: norm_from_string(pal): Bogus norm parameter, setting default.
v4l2: ioctl enum norm failed: Inappropriate ioctl for device
Error: Cannot set norm!
Selected input hasn't got a tuner!
v4l2: ioctl set mute failed: Invalid argument
Failed to open VDPAU backend libvdpau_vc4.so: cannot open shared object file: No such file or directory
[vdpau] Error when calling vdp_device_create_x11: 1
==========================================================================
Opening video decoder: [ffmpeg] FFmpeg's libavcodec codec family
libavcodec version 58.35.100 (external)
Selected video codec: [ffmjpeg] vfm: ffmpeg (FFmpeg MJPEG)
==========================================================================
Audio: no sound
Starting playback...
[mjpeg @ 0xb5d72878]No JPEG data found in image
Error while decoding frame!
Could not find matching colorspace - retrying with -vf scale...
Opening video filter: [scale]
Movie-Aspect is undefined - no prescaling applied.
[swscaler @ 0xb6839400]bicubic scaler, from yuv422p to yuv420p using C
VO: [xv] 640x480 => 640x480 Planar YV12 
Movie-Aspect is undefined - no prescaling applied.
VO: [xv] 640x480 => 640x480 Planar YV12 
V:   0.0 245/245 ??% ??% ??,?% 0 0 
No bind found for key 'CTRL-c'.
V:   0.0 334/334 ??% ??% ??,?% 0 0 

v4l2: ioctl set mute failed: Invalid argument
v4l2: 335 frames successfully processed, 26 frames dropped.

Exiting... (End of file)
```
vlc gives error: 
```
pi@raspberrypi:~/cam_overlay $ vlc v4l2:///dev/video0 -v
VLC media player 3.0.11 Vetinari (revision 3.0.11-0-gdc0c5ced72)
[01adab58] main libvlc: Running vlc with the default interface. Use 'cvlc' to use vlc without interface.
qt5ct: using qt5ct plugin
qt5ct: D-Bus global menu: no
qt5ct: D-Bus system tray: no
[99c00a28] v4l2 demux warning: cannot reset cropping limits: Inappropriate ioctl for device
[99c07618] mmal_codec decoder: VCSM init succeeded: CMA
[99c07618] main decoder error: buffer deadlock prevented
QObject::~QObject: Timers cannot be stopped from another thread

```

egisz, 2020-07-31:
@meekys , could it be related to RPI4 different implementation of OpenGL? I found some interesting info [here:](https://github.com/matusnovak/rpi-opengl-without-x#raspberry-pi-4)

meekys, 2020-08-01:
@egisz It's quite possible it's related the issue you've linked. cam_overlay is using `bcm_host_init` and vc related functions (`vc_dispmanx_display_open`, `vc_dispmanx_update_start', etc...) for direct display control.

I haven't got a pi4 to test with, so I couldn't say for certain.

If you have X11 running, you may be able to force it to use an X11 window instead (on the cpp branch) by using `--display=X11`. This would at least let us narrow down the issue.

The link you provided would probably be enough to get started on either changing (if still backwards compatibly with older pi's) or creating an alternative display option for the pi4

egisz, 2020-10-12:
@meekys , could you update instructions how to properly install prerequisites to build cpp branch?
And could you check if it builds for you?

I face 2 problems:
1) cmake is blaming it cannot find OPENGL_opengl_LIBRARY on both RPi3 and 4:

```
pi@CRANKSHAFT-NG:~/src/cam_overlay/src/Gl $ cmake . -LH
CMake Error at /usr/share/cmake-3.13/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find OpenGL (missing: OPENGL_opengl_LIBRARY)
Call Stack (most recent call first):
  /usr/share/cmake-3.13/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.13/Modules/FindOpenGL.cmake:395 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:21 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/pi/src/cam_overlay/src/Gl/CMakeFiles/CMakeOutput.log".
-- Cache values
// Choose the type of build, options are: None Debug Release RelWithDebInfo MinSizeRel ...
CMAKE_BUILD_TYPE:STRING=

// Install path prefix, prepended onto install directories.
CMAKE_INSTALL_PREFIX:PATH=/usr/local

// Path to a file.
OPENGL_EGL_INCLUDE_DIR:PATH=/usr/include

// Path to a file.
OPENGL_GLX_INCLUDE_DIR:PATH=/usr/include

// Path to a file.
OPENGL_INCLUDE_DIR:PATH=/usr/include

// Path to a library.
OPENGL_egl_LIBRARY:FILEPATH=/usr/lib/arm-linux-gnueabihf/libEGL.so

// Path to a library.
OPENGL_gl_LIBRARY:FILEPATH=/usr/lib/arm-linux-gnueabihf/libGL.so

// Path to a library.
OPENGL_glu_LIBRARY:FILEPATH=OPENGL_glu_LIBRARY-NOTFOUND

// Path to a library.
OPENGL_glx_LIBRARY:FILEPATH=/usr/lib/arm-linux-gnueabihf/libGLX.so

// Path to a library.
OPENGL_opengl_LIBRARY:FILEPATH=OPENGL_opengl_LIBRARY-NOTFOUND

// Path to a file.
OPENGL_xmesa_INCLUDE_DIR:PATH=OPENGL_xmesa_INCLUDE_DIR-NOTFOUND

```
If I modify src/Gl/CmakeLists.txt, and remove REQUIERED, then cmake succeds, but I'm not sure if this is OK or not? 
find_package(OpenGL REQUIRED COMPONENTS EGL)

2) make command fails:
```
Scanning dependencies of target DisplayBcm
[ 48%] Building CXX object src/Display/Bcm/CMakeFiles/DisplayBcm.dir/src/DisplayBcm.cpp.o
/home/pi/src/cam_overlay/src/Display/Bcm/src/DisplayBcm.cpp: In constructor ‘DisplayBcm::DisplayBcm(std::shared_ptr<ILogger>)’:
/home/pi/src/cam_overlay/src/Display/Bcm/src/DisplayBcm.cpp:10:10: error: ‘void init’ has incomplete type
     auto init = bcm_host_init();
          ^~~~
/home/pi/src/cam_overlay/src/Display/Bcm/src/DisplayBcm.cpp:12:23: error: unable to find string literal operator ‘operator""_’ with ‘const char [29]’, ‘unsigned int’ arguments
         _logger->Warn("bcm_host_init() failed with "_ + init);
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/src/cam_overlay/src/Display/Bcm/src/DisplayBcm.cpp:19:25: error: unable to find string literal operator ‘operator""_’ with ‘const char [41]’, ‘unsigned int’ arguments
         throw Exception("graphics_get_display_size() failed with "_ + success);
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/src/cam_overlay/src/Display/Bcm/src/DisplayBcm.cpp:19:15: error: ‘Exception’ was not declared in this scope
         throw Exception("graphics_get_display_size() failed with "_ + success);
               ^~~~~~~~~
make[2]: *** [src/Display/Bcm/CMakeFiles/DisplayBcm.dir/build.make:63: src/Display/Bcm/CMakeFiles/DisplayBcm.dir/src/DisplayBcm.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:316: src/Display/Bcm/CMakeFiles/DisplayBcm.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

```

egisz, 2020-10-12:
I have some small progress: 
I modified `cam_overlay/src/RaspberryPi.cmake`:
changed NAMES from brcmEGL to **lib**brcmEGL
```
  find_library(OPENGL_gl_LIBRARY
    NAMES libbrcmEGL
    PATHS /opt/vc/lib
    NO_DEFAULT_PATH
  )

  find_library(OPENGL_egl_LIBRARY
    NAMES libbrcmEGL
    PATHS /opt/vc/lib
    NO_DEFAULT_PATH
  )
```
Now cmake finds all dependencies:
```
pi@CRANKSHAFT-NG:~/src/cam_overlay $ cmake .
display_projects DisplayBcm;DisplayX11
input_projects InputV4L2;InputFile
output_projects OutputJpeg;OutputShader;OutputFile
display_modules Bcm;X11
input_modules V4L2MMap;V4L2UserPtr;V4L2Read;File
output_modules Jpeg;Shader;File
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/src/cam_overlay
```

but make still fails at same file:
```
[ 48%] Building CXX object src/Display/Bcm/CMakeFiles/DisplayBcm.dir/src/DisplayBcm.cpp.o
/home/pi/src/cam_overlay/src/Display/Bcm/src/DisplayBcm.cpp: In constructor ‘DisplayBcm::DisplayBcm(std::shared_ptr<ILogger>)’:
/home/pi/src/cam_overlay/src/Display/Bcm/src/DisplayBcm.cpp:10:10: error: ‘void init’ has incomplete type
     auto init = bcm_host_init();
          ^~~~
/home/pi/src/cam_overlay/src/Display/Bcm/src/DisplayBcm.cpp:12:23: error: unable to find string literal operator ‘operator""_’ with ‘const char [29]’, ‘unsigned int’ arguments
         _logger->Warn("bcm_host_init() failed with "_ + init);
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/src/cam_overlay/src/Display/Bcm/src/DisplayBcm.cpp:19:25: error: unable to find string literal operator ‘operator""_’ with ‘const char [41]’, ‘unsigned int’ arguments
         throw Exception("graphics_get_display_size() failed with "_ + success);
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/src/cam_overlay/src/Display/Bcm/src/DisplayBcm.cpp:19:15: error: ‘Exception’ was not declared in this scope
         throw Exception("graphics_get_display_size() failed with "_ + success);
               ^~~~~~~~~
make[2]: *** [src/Display/Bcm/CMakeFiles/DisplayBcm.dir/build.make:63: src/Display/Bcm/CMakeFiles/DisplayBcm.dir/src/DisplayBcm.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:316: src/Display/Bcm/CMakeFiles/DisplayBcm.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

meekys, 2020-10-15:
It's possible the libraries for Bmc have changed slightly in later versions of Raspbian/Raspberry Pi OS, as you've found with `brcmEGL` in `RaspberryPi.cmake`

On top of this, it looks like `bcm_host_init()` now returns void too, so it'd just be a matter of not assigning the result (as there now isn't one) to a variable and then not checking that variable.

In `src/Display/Bcm/src/DisplayBcm.cpp`
```
    auto init = bcm_host_init();
    if (init < 0)
        _logger->Warn("bcm_host_init() failed with "_ + init);
```
Replaced with
```
    bcm_host_init();
```

For the `unable to find string literal operator ‘operator""_’` type errors, you might need to replace the `_` with an `s`.

I have to admit, I haven't tried with the latest version(s) of Raspbian/Raspberry Pi OS, but if this fixes it to compile, it might be worth creating a pull request for these fixes.

Compizfox, 2021-06-03:
Any updates on this?

I'm experiencing the same issue on Crankshaft on a RPi 3A+:

```$ ./cam_overlay.bin
Opened device '/dev/video0'
Dimensions: 720 x 480 Pixel Format: YUYV
* failed to add service - already in use?
```