meekys/cam_overlay #28: * failed to add service - already in use?

Maxwime, 2022-04-14:
Hello,

First, thanks a lot for your work!
I'm a noob in raspberry system, so I need your help.

I'm trying to install a raspberry PI3 B+ with Crankshaft, but the REAR CAM doses not work.
I search a lot to understand why, and I search finish on cam_overlay....

But when I try to launch it with:

. /cam_overlay.bin

I get:

Opened device '/dev/video0'
Dimensions: 352 x 288 Pixel Format: MJPG
* failed to add service - already in use?

Did I do something wrong?

I hope my message is in the good place...

Thanks

meekys, 2022-04-14:
Hi Maxwime,

`cam_overlay` does not currently support `dtoverlay=vc4-kms-v3d`, so if you don't need this feature, you can remove it from `config.txt`

I'm just in the process of getting some PI hardware again to experiment further with this.

https://raspberrypi.stackexchange.com/questions/75219/fixing-failed-to-add-service-already-in-use-error-programmatically/75688

Maxwime, 2022-04-18:
Thanks for tour return,
I already try to remove
Dtoverlay=vc4-kms-v3d
But that makes Crankshaft crash.
I will continue to search with the link tout give me.
If I found a solution I will come back to tell you.
Thanks again

Maxwime, 2022-05-09:
Hello, I'm back :)

I found a way to make it work.

I install "[Crankshaft OpenCarOS NG Alpha7.2 2021-02-09]" version on my PI3B+
Then I make it start in X11 mode
After i comment in the "config.txt":

[all]
dtoverlay=vc4-fkms-v3d

and then everything is all right

thank

closed 2022-05-09 by Maxwime

kadapppa, 2023-05-06:
I am clue less for this error : * failed to add service - already in use?

Got keys from plugin meta data ("eglfs_x11")
QFactoryLoader::QFactoryLoader() checking directory path "/home/root/kp/egldeviceintegrations" ...
qt.qpa.egldeviceintegration: EGL device integration plugin keys: ("eglfs_emu", "eglfs_x11")
qt.qpa.egldeviceintegration: Using base device integration
* failed to add service - already in use?
QLibraryPrivate::unload succeeded on "/usr/lib/plugins/platforms/libqeglfs.so"
root@raspberrypi3:~/kp#

any hint ?