meekys/cam_overlay #29: Crankshaft black screen

Ramortada, 2023-02-21:
Hello,

When connecting a USB webcam with crankshaft, I get a black screen whenever I switch to the camera.

The device supports MJPG pixel format, I don't know if this is the cause. I have similar results with this issue. https://github.com/meekys/cam_overlay/issues/10

I would very much appreciate any help

meekys, 2023-02-21:
@Ramortada Just a couple of questions.
- Which branch are you using? MJPG is only available in the cpp branch
- After stopping (Ctrl+C or killing the process), what is logged in the terminal?

Ramortada, 2023-02-23:
I don't know what you mean by cpp branch. I am using an FPV RCA camera connected to usb with this device : https://fr.aliexpress.com/item/1005004107556241.html?spm=a2g0o.cart.0.0.3f9e378dy4kvvB&mp=1&gatewayAdapt=glo2fra

I get a video signal on my laptop.
Could be the same as issue #25 ?
 
I won't have access to my setup until next week, I can try any commands Im sorry

meekys, 2023-02-23:
@Ramortada Without seeing the console output, I can't really say why it might not be working. Once I see them, hopefully we can figure out what's causing the black screen.

Did you compile `cam_overlay` yourself, or are you using it as part of `crankshaft`?

Ramortada, 2023-02-23:
Thank you this is very nice of you. 
What do you mean by seeing the console output? 
And I am using `cam_overlay`  with `crankshaft`

Ramortada, 2023-03-01:
Hello, so I have the output when I use 'crankshaft rearcam show : 
`Opened device ' /dev/video0'
Dimensions : 640 x 480 Pixel Format : MJPG
Cannot open 'shader-MJPG.frag' : 2, No such file or directory` 

I get the same result if the cam is connected and not connected. I also tried to remove the 'dtoverlay=vc4-fkms-v3d' command from config.txt but I had the same output.

meekys, 2023-03-03:
Ok, thanks for the logs.

This is an issue with `crankshaft`, as they'll need to update to the `cpp` branch, which should support `MJPG`

Unfortunately, as there are breaking changes, this hasn't happened yet

meekys, 2023-03-03:
I'm not sure, sorry. I'm not familiar enough with how it's been integrated into crankshaft

The best I can suggest is looking at `stage3/03-crankshaft-base/files/opt/crankshaft/service_gpiotrigger.sh`, as that looks like it starts the process.

You'd need to compile the `cpp` branch of `cam_overlay`, as well as integrating it into `crankshaft` in that script (presumably)

Ramortada, 2023-03-03:
I appreciate it, I will start from there

Geekyadz, 2024-05-26:
Hi @Ramortada 

Did you have any joy with this?

I'm kinda in the same boat, certainly in the sense of I'm getting:

Crankshaft Management Tool v2.4.0 is executing command...

Show rearcam...

Done.
pi@CRANKSHAFT-NG:~ $ Opened device '/dev/video0'
Dimensions: 1920 x 1080 Pixel Format: MJPG
* failed to add service - already in use?

Any suggestions on how to resolve this issue?