meekys/cam_overlay #31: Another Crankshaft related issue about cam_overlay on Pi 3 model B and Pi 4B

Geekyadz, 2024-06-16:
Hi,

I have checked most of the issues here and after spending most of all day on this, I feel like I need some help.

I'm trying to use this ( https://www.avsl.com/product/128.836UK-4k-hdmi-video-capture-card ) with a RCA to HDMI converter so I can then look into what reversing camera I can go and get setup with. Not to mention I am using a USB extension as it's too big to fit into the Pi (I'm kinda ok with this for the moment - also I have a RCA to HDMI convertor in case you're wondering that a reversing camera I am borrowing works with).

Let's start with the Pi 3 Model B, cos this is doing what I want (but not quite if you know what I mean) and feel like it's going to be "the easier fix". For context, I am using SSH on my desktop to try and make this easier on myself:

Initially I noticed that I had the issue with issue #29 (specifically, https://github.com/meekys/cam_overlay/issues/29#issuecomment-1450208677 ) so I commented out the dtoverlay=vc4-fkms-v3d line for the Pi 3, then I was also stuck with the "Cannot open 'shader-MJPG.frag'" error.

In the name of doing more testing, I installed FFMPEG and ran `ffplay /dev/video0` which displayed part of the reversing camera I am currently borrowing for testing purposes. Then I decided to try and run cam_overlay.bin again, and this time it displayed in full screen exactly what I would like.

I've therefore done a little more digging, and found that the capture card has two modes as seen below:

[image]

The final piece of this puzzle, then it's back to being a "me" problem: is there a way of telling/setting the pixel format of V4L2 to use the YUYV format instead of trying to grab the MJPG format? If so, can someone please provide the command to use please?

As for the Pi 4, well, I think the easiest way is for me to say what needs to be done I feel as I'm getting the same error whether I comment out the line above from earlier or not (and getting the dreaded "failed to add service - already in use?" when trying to run cam_overlay.bin

Download Crankshaft using git in SSH
Download CCP branch of cam_overlay
Make/build both, then test again

Issue lies: I'm having difficulty with the git command and building in general, but I would like a list of commands please I am meant to be using, just to make sure I am not making any "newbie" mistakes.

Thankyou for reading this essay. In both cases, please feel free to let me know if you require further information; I'm bound to have forgotten something

meekys, 2024-06-20:
It's been a while since I've looked at these problems, but let me try to summarise:
- Commenting out `dtoverlay=vc4-fkms-v3d` will free up the device (on a pi3 at least) to allow camoverlay to use the video hardware directly
- On a pi4, I believe there is a different video architecture/driver, which I haven't looked into yet. This is probably why the pi4 does not work.
-To set the default format, try running something like this, `v4l2-ctl --device path/to/video_device --set-fmt-video=width=width,height=height,pixelformat=format`. Running `man v4l2-ctl` should get you all the options available.
- The cpp branch should support the MJPEG format, but has a different command line, so might not be completely compatible with crankshaft. I'm not that familiar with the crankshaft build pipeline, but to checkout the cpp branch of camoverlay, run `git checkout cpp` from within the source directory, then try follow the `README.md` in the same directory to compile.
- If you're having any specific build issues, you'll need to copy/paste the output to narrow down the cause.

Geekyadz, 2024-06-20:
> It's been a while since I've looked at these problems, but let me try to summarise:
>
> * Commenting out `dtoverlay=vc4-fkms-v3d` will free up the device (on a pi3 at least) to allow camoverlay to use [...]

Thankyou for your response. 

Unfortunately the Pi3 wasn't mine and at the moment I don't have access to it, though as I've said above: if this is going to be quicker to get working I'll happily get that for now as a work around. I'll see if I can grab the Pi3 for next week and see how I get on with it with setting the format to the non MPEG one as that sounds easier than using MPEG at this stage. 

Speaking of MPEG and the CPP branch of cam overlay: is the a way of uninstalling/removing the current version of cam overlay from crankshaft? I'm just thinking that it might be easier for it to be removed then we can add it manually if required if we are having to switch the branch that it's currently on?

I'm also tempted to setup my pi4 so you can either VNC and/or SSH into it for you to check things if this assists you? I mean, I'm not the first person to have this issue and rather than everyone coming to this GitHub and contacting you about it, I'd rather try to stop this from being a thing?

Geekyadz, 2024-06-22:
>   -To set the default format, try running something like this, `v4l2-ctl --device path/to/video_device --set-fmt-video=width=width,height=height,pixelformat=format`. Running `man v4l2-ctl` should get you all the options available.
Just to provide an update: I have brought my own Raspberry Pi3 Model B and just ran this command on it:

`v4l2-ctl --device /dev/video0 --set-fmt-video=pixelformat=YUYV`

That worked, thank you for this.

I'll try building the CPP branch of camoverlay on my "main" Pi and let you know how I get on

Geekyadz, 2024-06-23:
Update for you: https://youtu.be/QF2X8LUEzLY

[Or if you prefer reading](https://github.com/opencardev/crankshaft/discussions/659#discussioncomment-9852820)

Again, thank you for your help

PointyFinger, 2024-11-01:
Is there a way to get this working for RPI 4B? I get the 'failed to add service - already in use" error and it's driving me crazy. Any help would be appreciated!